felias-fogg/SoftI2CMaster #4: Compilre hangs in Arduino 1.6.4 IDE

MarcoBlulab, 2015-09-13:
Hello,
I tried to use the library on Arduino 1.6.4 (Arduino Nano @ 16MHz)
I can't compile project. The compiler hangs (freeze) and never ends compilation (even after hours). I've tried several sketches, but compiler hangs out simply by including the library ".h" file.
Any suggestions?

```c
#include <SoftI2CMaster.h>
void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
}
```
Compiler output (verbose)
```
Uso la libreria SoftI2CMaster nella cartella: D:\prj\arduino\libraries\SoftI2CMaster (1.0.x format)

D:\prj\arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -Wall -Wextra -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -ID:\prj\arduino\hardware\arduino\avr\cores\arduino -ID:\prj\arduino\hardware\arduino\avr\variants\eightanaloginputs -ID:\prj\arduino\libraries\SoftI2CMaster C:\Users\MARCO_~1\AppData\Local\Temp\build6310432117920169458.tmp\sketch_sep13a.cpp -o C:\Users\MARCO_~1\AppData\Local\Temp\build6310432117920169458.tmp\sketch_sep13a.cpp.o 
```

brownjar, 2015-09-23:
I'm having the same issue...none of the example will even compile:

C:\Program Files (x86)\Arduino/hardware/tools/arm/bin/arm-none-eabi-g++ -c -O -g -Wall -ffunction-sections -fdata-sections -MMD -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -D__MK20DX256__ -DTEENSYDUINO=123 -DARDUINO=10604 -DF_CPU=96000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL -DLAYOUT_US_ENGLISH -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3 -IC:\Users\Travis\Documents\Arduino\libraries\SoftI2CMaster C:\Users\Travis\AppData\Local\Temp\build7501985190473441758.tmp\I2CScanSoft.cpp -o C:\Users\Travis\AppData\Local\Temp\build7501985190473441758.tmp\I2CScanSoft.cpp.o

felias-fogg, 2015-09-23:
Running 1.6.5, I cannot reproduce the problems (for the UNO platform).

brownjar, 2015-10-01:
UNO works for me too. And Teensy 2.0 works. But the platform I need, Teensy 3.1 does not work. I noticed Teensy 3.1 uses a different compiler and slightly different include files.

Teensy 2.0:
```
C:\Program Files (x86)\Arduino/hardware/tools/avrteensy/bin/avr-g++ ...
... -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy
```
Teensy 3.1:
```
C:\Program Files (x86)\Arduino/hardware/tools/arm/bin/arm-none-eabi-g++ ...
...-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3
```

How can I get Teensy 3.1 supported?

steveheinsch, 2016-03-01:
I would love Teensy 3.1 support as well, if possible.

felias-fogg, 2016-03-01:
Since the Teensy3.x processor is not an AVR processor, there won't be any port soon. However, there are some Arduino libs out there that implement software I2C.

closed 2016-09-29 by felias-fogg