felias-fogg/SoftI2CMaster #6: Is there a way to access multiple devices using your library?

hexibot43, 2015-11-27:
Your library is working quite well with my ArduinoMega and a single MLX90614.  I thought I would be able to access multiple devices but realized I had to "#define DEVICE" which gives me but one choice.  I have two MLX90614 tied to the same two pins for SCL and SCA.  I have changed the address of one of them.  Would there be an easy way to modify your code to work with more than one DEVICE? 

Thanks,  

Marcello

felias-fogg, 2015-11-28:
Hi Marcello,

you can use as many I2C devices on the I2C bus as you want, provided they have different I2C addresses. In the example sketch, you simply use different addresses when addressing the device. This is done in lines 30 and 33 in the example sketch MLX90614Soft. By the way, the constant DEVICE is never used (my fault). Instead I use the variable dev. Here you could use, e.g. dev1 and dev2 and assign different device addresses.

Cheers
Bernhard

hexibot43, 2015-11-28:
Bernhard you are the Man!  Thank you.  My project is now reading two MLX90614 devices using your library.  I hope you remove the "#Define Device" to make it more obvious for others that happen upon your work.  You've made my day.  Many Thanks,  Marcello  - Is there a way to have multipe I2C buses set up at the same time?  Instead of defining the SDA, SCK you pass them to your code.  "I2c_Init(SDA, SCK );"   That way if you had two or more devices with the same address you couldn't change?  Thanks,  Marcello  - I really appreciate the reply.

closed 2016-09-29 by felias-fogg