felias-fogg/SoftI2CMaster #9: Help understanding how to read/write to a specific register address

NCharabaruk, 2016-05-04:
Hello,

I am trying to put together an .ino to use SoftI2CMaster with an MPU6050 IMU but am somewhat confused. Can SoftI2CMaster write to a specific register? For example, if I want to configure the accelerometer on the IMU to use a range of +-4g I need to send a byte (00001000) to register 0x1C at bus address 0x68. From the examples here it looks like I can specify the bus address and the value of the byte to send, but I don't see how to specify the register address. As well, I also need to figure out how to read from a specific register address.

Any help you can give me would be greatly appreciated.

Thanks

eadf, 2016-05-04:
Hello again! (seems like the Arduino i2c world is small)

To write`length`bytes to a register address you can do something like this:
```
i2c_start(devAddr<<1 | I2C_WRITE);
i2c_write(regAddr);
for (uint8_t i = 0; i < length; i++) {
  i2c_write((uint8_t) data[i]);
}
i2c_stop();
```

To read`length`number of bytes from a register you can do this: 
```
if (!i2c_start(devAddr<<1 | I2C_WRITE)) return -1;
if (!i2c_write(regAddr)) return -1;
if (!i2c_rep_start(devAddr<<1 | I2C_READ)) return -1;
for (; count < length && (timeout == 0 || millis() - t1 < timeout);) {
    data[count] = i2c_read(count < length -1?false:true);
    count++;
}
i2c_stop();
```

This is just quickly copy&pasted from a [ i2cdevlib PR ](https://github.com/eadf/i2cdevlib/tree/SoftI2CMaster)

NCharabaruk, 2016-05-04:
Hello! I cross posted it. I figured if I could get one device working with the other's I2C library I could combine them on one board. This is not the type of programming I am used to though so I got stuck both ways :P

Thanks for all the help!

closed 2016-05-04 by NCharabaruk

eadf, 2016-05-05:
> I am trying to put together an .ino to use SoftI2CMaster with an MPU6050 IMU

I forgot to tell you this, but the same Pull Request I was copy & pasting from contains a SoftI2CMaster backend for i2cdevlib. So all* of the drivers (including MPU6050) of i2cdevlib can be used with SoftI2CMaster as the backbone. 
*I'm still testing the few i2c devices as I have access to, but MPU6050 IMU works without any problems so far.

NCharabaruk, 2016-05-05:
Does it work with DMP?

eadf, 2016-05-05:
Yes, DMP6 works. 
Just make sure that you flip the I2CDEV_IMPLEMENTATION #define in I2Cdev.h before use.
```
// -----------------------------------------------------------------------------
// I2C interface implementation setting
// -----------------------------------------------------------------------------
#ifndef I2CDEV_IMPLEMENTATION
//#define I2CDEV_IMPLEMENTATION       I2CDEV_ARDUINO_WIRE
//#define I2CDEV_IMPLEMENTATION       I2CDEV_BUILTIN_NBWIRE
//#define I2CDEV_IMPLEMENTATION       I2CDEV_BUILTIN_FASTWIRE
//#define I2CDEV_IMPLEMENTATION       I2CDEV_I2CMASTER_LIBRARY
#define I2CDEV_IMPLEMENTATION       I2CDEV_SOFTI2CMASTER_LIBRARY
#endif // I2CDEV_IMPLEMENTATION
```
Use the [i2cdevlib PR page](https://github.com/jrowberg/i2cdevlib/pull/220) if you need further assistance

NCharabaruk, 2016-05-05:
Ok, I will try putting the MB1222 code in there and see if I can get both sensors working