felias-fogg/SoftI2CMaster #10: Changes to run multiple instances on different pairs of pins?

Humancell, 2016-06-28:
Thank you for the great library ... it works as advertised!  :-)

I'm now looking at a situation where I need to support multiple I2C buses ... so I need multiple "masters".  Have you at all considered what changes would have to be made to support multiple instances?  What could remain shared, and what has to be renamed, or changed in definitions?

Looking at the code, I can see it's not trivial ... however I need to make this work.  I've started to review the code and attempt to determine all of the places that would require changes.  I am also not sure if there is an easy way to define "arrays" of pins ... or some other method that might be the best direction to attack this?

Thoughts?  Suggestions?

felias-fogg, 2016-06-28:
Hi Scott,

I wrote the lib mainly to be a very light weight solution for ATtinys which run on a 1MHz clock. So it it uses the minimal amount of code and all the ports are hardcoded (through defines). Lifting this to multiple buses is probably very difficult. For two buses, I would just duplicate the code with different identifiers.

But why would you like to do it? Since everything is based on polling the I/O lines, there won’t be any advantage for serving parallel buses. Do you have chips that have identical I2C addresses that you cannot change?

If you have more memory to spare, you might probably want to play around with the high-level libraries implementing I2C in software. I believe, they also do not support multiple buses, but it might be easier to extend them to cater for that.

Best, Bernhard

closed 2016-09-29 by felias-fogg

MarkJeronimus, 2020-10-26:
I am looking at this possibility too. I have to control 8 identical I2C modules with fixed address. Searching, I found this. After trying and failing, I found your SlowSoftI2CMaster (which seems to accept two GPIO parameters in a constructor). I will now test that.