felias-fogg/SoftI2CMaster #12: cannot compile example // referenced in section `.text' of /var/folders

marioappinger, 2016-08-09:
i guess it is not really an issue about your code!
i just wanted to compile the example "i2cscansoft" under "Arduino IDE 1.6.10 for mac"
sorry to ask - but i do not realy understand the error messages below
have you any idea what that should mean?

KR from Vienna
Mario

'ass_i2c_delay_half' referenced in section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T//ccxTt4XP.ltrans0.ltrans.o: defined in discarded section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T/build269360e7d88ea95ed2149f23453ce0d2.tmp/sketch/softi2c_testappi.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T//ccxTt4XP.ltrans0.ltrans.o: defined in discarded section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T/build269360e7d88ea95ed2149f23453ce0d2.tmp/sketch/softi2c_testappi.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T//ccxTt4XP.ltrans0.ltrans.o: defined in discarded section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T/build269360e7d88ea95ed2149f23453ce0d2.tmp/sketch/softi2c_testappi.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T//ccxTt4XP.ltrans0.ltrans.o: defined in discarded section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T/build269360e7d88ea95ed2149f23453ce0d2.tmp/sketch/softi2c_testappi.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T//ccxTt4XP.ltrans0.ltrans.o: defined in discarded section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T/build269360e7d88ea95ed2149f23453ce0d2.tmp/sketch/softi2c_testappi.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T//ccxTt4XP.ltrans0.ltrans.o: defined in discarded section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T/build269360e7d88ea95ed2149f23453ce0d2.tmp/sketch/softi2c_testappi.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T//ccxTt4XP.ltrans0.ltrans.o: defined in discarded section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T/build269360e7d88ea95ed2149f23453ce0d2.tmp/sketch/softi2c_testappi.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T//ccxTt4XP.ltrans0.ltrans.o: defined in discarded section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T/build269360e7d88ea95ed2149f23453ce0d2.tmp/sketch/softi2c_testappi.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T//ccxTt4XP.ltrans0.ltrans.o: defined in discarded section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T/build269360e7d88ea95ed2149f23453ce0d2.tmp/sketch/softi2c_testappi.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T//ccxTt4XP.ltrans0.ltrans.o: defined in discarded section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T/build269360e7d88ea95ed2149f23453ce0d2.tmp/sketch/softi2c_testappi.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T//ccxTt4XP.ltrans0.ltrans.o: defined in discarded section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T/build269360e7d88ea95ed2149f23453ce0d2.tmp/sketch/softi2c_testappi.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T//ccxTt4XP.ltrans0.ltrans.o: defined in discarded section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T/build269360e7d88ea95ed2149f23453ce0d2.tmp/sketch/softi2c_testappi.ino.cpp.o (symbol from plugin)
`ass_i2c_write' referenced in section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T//ccxTt4XP.ltrans0.ltrans.o: defined in discarded section `.text' of /var/folders/nb/z09lfwz50tq43_330j0lyvrm0000gn/T/build269360e7d88ea95ed2149f23453ce0d2.tmp/sketch/softi2c_testappi.ino.cpp.o (symbol from plugin)
collect2: error: ld returned 1 exit status
Bibliothek SoftI2CMaster-master im Ordner: /Users/map/Documents/Arduino/libraries/SoftI2CMaster-master (legacy) wird verwendet
exit status 1

townsweb, 2016-08-14:
I'm having the same problem. I'm running Linux I've tried arduino 1.6.9 and 1.6.10

`ass_i2c_wait_scl_high' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Uno.

SpicerSolutions, 2016-08-14:
I'm having the same problem, I'm running Linux, I've tried the code on both 1.6.9 and 1.6.10.

`ass_i2c_wait_scl_high' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_wait_scl_high' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
`ass_i2c_delay_half' referenced in section `.text' of /tmp/cc3QQgX6.ltrans0.ltrans.o: defined in discarded section `.text' of sketch/mag3110_v10.ino.cpp.o (symbol from plugin)
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Uno.

SpicerSolutions, 2016-08-14:
So after a little investigation it appears to be a bug with the compiler (between gcc-4.7 and gcc-4.8).

A work around is to change the following lines so they are not  discarded.

void __attribute__ ((noinline)) i2c_stop(void) asm("ass_i2c_stop");
bool __attribute__ ((noinline)) i2c_write(uint8_t value) asm("ass_i2c_write");
void __attribute__ ((noinline)) i2c_delay_half(void) asm("ass_i2c_delay_half");
void __attribute__ ((noinline)) i2c_wait_scl_high(void) asm("ass_i2c_wait_scl_high");

becomes

void __attribute__ ((noinline)) i2c_stop(void) asm("ass_i2c_stop")  __attribute__((used));
bool __attribute__ ((noinline)) i2c_write(uint8_t value) asm("ass_i2c_write")  __attribute__((used));
void __attribute__ ((noinline)) i2c_delay_half(void) asm("ass_i2c_delay_half")  __attribute__((used));
void __attribute__ ((noinline)) i2c_wait_scl_high(void) asm("ass_i2c_wait_scl_high")  __attribute__((used));

felias-fogg, 2016-08-14:
Thanks! Will change the code when I am back home.

Per-Thulin, 2016-09-07:
change 2 instances of ((noinline)) into: ((used,noinline))

coll0412, 2016-09-17:
I used Per-Thulin recommendation and modified the following 4 lines.  SpicerSolutions method would not compile in Arduino 1.6.11

void __attribute__ ((used,noinline)) i2c_stop(void) asm("ass_i2c_stop");
bool __attribute__ ((used,noinline)) i2c_write(uint8_t value) asm("ass_i2c_write");
void __attribute__ ((used,noinline)) i2c_delay_half(void) asm("ass_i2c_delay_half");
void __attribute__ ((used,noinline)) i2c_wait_scl_high(void) asm("ass_i2c_wait_scl_high");

weicher, 2016-09-21:
I tried SpicerSolutions and Coll0412. I got nearly the same error messages:

SpicerSolutions:
SoftI2CMaster.h: 86:16: error: variable or field 'attribute' declared void
   void attribute ((noinline)) i2c_stop(void) asm("ass_i2c_stop") attribute((used))
 
SoftI2CMaster.h: 86:18: error: 'noinline' was not declared in this scope
   void attribute ((noinline)) i2c_stop(void) asm("ass_i2c_stop") attribute((used))
 
SoftI2CMaster.h: 91:18: error: 'noinline' was not declared in this scope
   bool attribute ((noinline)) i2c_write(uint8_t value) asm("ass_i2c_write") attribute((used))
 
SoftI2CMaster.h: 91:29: error: expected ',' or ';' before 'i2c_write
   bool attribute ((noinline)) i2c_write(uint8_t value) asm("ass_i2c_write") attribute((used))
 
SoftI2CMaster.h: 182:16: error: variable or field 'attribute' declared void
   void attribute ((noinline)) i2c_delay_half(void) asm("ass_i2c_delay_half") attribute((used))
 
SoftI2CMaster.h: 182:18: error: 'noinline' was not declared in this scope
   void attribute ((noinline)) i2c_delay_half(void) asm("ass_i2c_delay_half") attribute((used))
 
SoftI2CMaster.h: 183:16: error: variable or field 'attribute' declared void
   void attribute ((noinline)) i2c_wait_scl_high(void) asm("ass_i2c_wait_scl_high") attribute((used))
 
SoftI2CMaster.h: 183:18: error: 'noinline' was not declared in this scope
   void attribute ((noinline)) i2c_wait_scl_high(void) asm("ass_i2c_wait_scl_high") attribute((used))
Error compiling project sources

Coll0412:
SoftI2CMaster.h: 86:16: error: variable or field 'attribute' declared void
   void attribute ((used,noinline)) i2c_stop(void) asm("ass_i2c_stop")
 
SoftI2CMaster.h: 86:18: error: 'used' was not declared in this scope
   void attribute ((used,noinline)) i2c_stop(void) asm("ass_i2c_stop")
 
SoftI2CMaster.h: 86:23: error: 'noinline' was not declared in this scope
   void attribute ((used,noinline)) i2c_stop(void) asm("ass_i2c_stop")
 
SoftI2CMaster.h: 91:18: error: 'used' was not declared in this scope
   bool attribute ((used,noinline)) i2c_write(uint8_t value) asm("ass_i2c_write")
 
SoftI2CMaster.h: 91:23: error: 'noinline' was not declared in this scope
   bool attribute ((used,noinline)) i2c_write(uint8_t value) asm("ass_i2c_write")
 
SoftI2CMaster.h: 91:34: error: expected ',' or ';' before 'i2c_write
   bool attribute ((used,noinline)) i2c_write(uint8_t value) asm("ass_i2c_write")
 
SoftI2CMaster.h: 182:16: error: variable or field 'attribute' declared void
   void attribute ((used,noinline)) i2c_delay_half(void) asm("ass_i2c_delay_half")
 
SoftI2CMaster.h: 182:18: error: 'used' was not declared in this scope
   void attribute ((used,noinline)) i2c_delay_half(void) asm("ass_i2c_delay_half")
 
SoftI2CMaster.h: 182:23: error: 'noinline' was not declared in this scope
   void attribute ((used,noinline)) i2c_delay_half(void) asm("ass_i2c_delay_half")
 
SoftI2CMaster.h: 183:16: error: variable or field 'attribute' declared void
   void attribute ((used,noinline)) i2c_wait_scl_high(void) asm("ass_i2c_wait_scl_high")
 
SoftI2CMaster.h: 183:18: error: 'used' was not declared in this scope
   void attribute ((used,noinline)) i2c_wait_scl_high(void) asm("ass_i2c_wait_scl_high")
 
SoftI2CMaster.h: 183:23: error: 'noinline' was not declared in this scope
   void attribute ((used,noinline)) i2c_wait_scl_high(void) asm("ass_i2c_wait_scl_high")
Error compiling project sources

felias-fogg, 2016-09-29:
Now it is working again for the recent Arduino versions

closed 2016-09-29 by felias-fogg

josejuansanchez, 2016-10-07:
@felias-fogg Thanks! :)

ReverseEMF, 2018-03-22:
I'm having a similar problem in IDE 1.8.3 with the SoftI2CMaster module.  And adding **attribute((used))** causes a different compile error (i.e. doesn't fix the problem):

> In file included from C:\Users\Steve\AppData\Local\Temp\arduino_build_875801\sketch\max30102.cpp:63:0:
> 
> SoftI2CMaster.h:110: error: expected initializer before 'attribute'
> 
>  void __attribute__ ((noinline)) i2c_stop(void) asm("ass_i2c_stop") attribute((used));

Here's what the original errors look like:

> C:\Users\Steve\AppData\Local\Temp\ccBaeGrg.ltrans0.ltrans.o: In function `i2c_rep_start(unsigned char) [clone .constprop.9]':
> 
> C:\Users\Steve\AppData\Local\Temp\arduino_build_875801\sketch/SoftI2CMaster.h:336: undefined reference to `ass_i2c_delay_half'
> 
> C:\Users\Steve\AppData\Local\Temp\arduino_build_875801\sketch/SoftI2CMaster.h:336: undefined reference to `ass_i2c_delay_half'
> 
> C:\Users\Steve\AppData\Local\Temp\arduino_build_875801\sketch/SoftI2CMaster.h:336: undefined reference to `ass_i2c_delay_half'
> 
> C:\Users\Steve\AppData\Local\Temp\arduino_build_875801\sketch/SoftI2CMaster.h:336: undefined reference to `ass_i2c_wait_scl_high'
> 
> C:\Users\Steve\AppData\Local\Temp\arduino_build_875801\sketch/SoftI2CMaster.h:336: undefined reference to `ass_i2c_delay_half'
> etc...

felias-fogg, 2018-03-22:
Which version of the lib are you using? The current one is 2.1 and includes already all the \_\_attribute\_\_ (note the double underscores!) declarations that are needed.

fragsalat, 2019-02-11:
I'm having the same issue with latest master. Using gcc-avr 5.4.0+Atmel3.6.1-1~rpt1 and Arduino.
I want to build for an ATtiny84 but that should not matter right now. 

```
#define SDA_PORT PORTA
#define SDA_PIN 4 // = A4
#define SCL_PORT PORTA
#define SCL_PIN 5 // = A5

#include <SoftWire.h>

SoftWire TinyWireM = SoftWire();
```

```
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `ass_i2c_delay_half'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `ass_i2c_wait_scl_high'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `i2c_init()'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `i2c_start(unsigned char)'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `ass_i2c_write'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `i2c_rep_start(unsigned char)'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `i2c_start_wait(unsigned char)'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `ass_i2c_stop'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `i2c_read(bool)'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `TinyWireM'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
collect2: error: ld returned 1 exit status
```