felias-fogg/SoftI2CMaster #13: wrong address with i2c_start in write mode

JustMy2Cents, 2016-09-12:
Hi,
I'm using 1.6.11 and wanted to try out the soft I2C code.
The problem is as soon as I clear the read/write bit in i2c_start() the address data gets corrupted

correct address 0b0101001 with hardware I2C. Also works fine when read7write bis is set in read mode.
[image]

No matter which address I set, as soon as the read/write bit is cleared the address gets corrupted and always looks like this:
[image]

felias-fogg, 2016-09-29:
I am surprised that you got the program working under 1.16.11 since everybody else was not able to compile it. In any case, please post the program that leads to the wrong behavior.

JustMy2Cents, 2016-10-04:
Sorry for the delay, I was on vacation.
Here is the code:
[SoftI2CMaster.zip](https://github.com/felias-fogg/SoftI2CMaster/files/508062/SoftI2CMaster.zip)

felias-fogg, 2016-10-04:
Sorry, the file you sent me is just my library. What I asked for was your code that generates the problem you are showing. 

Best regards,
Bernhard

JustMy2Cents, 2016-10-04:
Hi Bernhard,
I use your code in combination with a proximity sensor:
[VL6180XSoft.zip](https://github.com/felias-fogg/SoftI2CMaster/files/508677/VL6180XSoft.zip)

felias-fogg, 2016-10-04:
Hi,

as far as I can see you use different I2C-addresses in the i2c_start and i2c_rep_start calls in readReg, namely 0b01011010 and 0b01010010. This does not look right. The only difference should be the R/W bit. Otherwise I don’t see a problem. 

If you could boil it down to a short sequence of calls with accompanying pictures of a logic analyzer or scope, I could try to reproduce it here and try to figure out, what went wrong.

Best,
Bernhard

P.S.: Note that I have updated the library and that there is now a Wire-like interface available.

felias-fogg, 2016-12-10:
I presume the problem has been solved or it is not relevant anymore. Therefore, I close the thread.

closed 2016-12-10 by felias-fogg