felias-fogg/SoftI2CMaster #16: Support for ATtiny841?

nadavmatalon, 2016-12-01:
Hi   Bernhard,

First of all, many thanks for writing this great library! it's been very helpful in a project I'm currently working on 👍 

I've successfully implemented your library on the ATtiny84A as an I2C bus master, but when I considered switching to the more advanced ATtiny841 I've run into trouble. More Specifically, in my project I'm running 2 completely separate I2C buses and the ATtiny is a member of both. On the first bus, the ATtiny is simply a slave device (using the TinyWireS library and the hw TWI pins, i.e. SDA: A6 & SCL: A4). On the second I2C bus, the ATtiny acts as Master and here is where your library comes in (using software I2C with: SDA: A0 & SCL: A5) . 

Like I said, when this setup is implemented on the ATtiny84A, it works without a hitch. However, switching to the ATtiny841, I discovered that running the 'Wire.begin()' function to initialize the Tiny as an I2C master causes its slave-side to stop working. I suspect the initialization process of the master-side combined with the different implementation of the TWI module on the ATtiny841 creates some kind of 'overriding' problem of registers or something similar. 

Although the two chips are pin compatible, they do have significant differences in memory mapping (apparently the RAM in the ATtiny84A is mapped to 0x0060, while in the ATtiny841 it is mapped to 0x100). 

I furthermore suspect that the issue described above may be related to these lines of your library:
```
#define SDA_DDR         (_SFR_IO_ADDR(SDA_PORT) - 1)   
#define SCL_DDR         (_SFR_IO_ADDR(SCL_PORT) - 1)   
#define SDA_OUT         _SFR_IO_ADDR(SDA_PORT)   
#define SCL_OUT         _SFR_IO_ADDR(SCL_PORT)   
#define SDA_IN      (_SFR_IO_ADDR(SDA_PORT) - 2)   
#define SCL_IN      (_SFR_IO_ADDR(SCL_PORT) - 2)   
```
If this is indeed so, do you have any suggestions as to how I might modify these lines to make my code run on the ATtiny 841? or, if you think it might be something else, could you enlighten me please? (btw, before writing all this, I've tried both Googling and looking at the datasheet for quite some time, but found no answer as to how the values in these lines are determined or what other cause might be involved).

If any further clarification/code example is needed, I'd of course be happy to provide it.
Hope you might be able to help.

All the best,
Nadav

felias-fogg, 2016-12-02:
I do not have an ATtiny 841 lying around, so I cannot check myself. 

Well, when you look into the assembler code of i2c_init, you see that only the data direction register bits and the data register bits are changed (on a per bit basis). So I cannot see that this can influence the working of the TWI part, which works on different pins. Similarly, the addressing scheme you mentioned with the "defines" should work with the 841 since the I/O registers have the same order on both MCUs (see the Register Summary sections in both data sheets). In other words, I do not have a good explanation. 

Do the two TWI modules work in isolation without problems?

nadavmatalon, 2016-12-05:
Hi again,

Many thanks for getting back to me on this and my sincere apologies... you are right and, as far as I can tell, the library works fine with the ATtiny841 :-)

long story short, I must have broken something small when playing around with your library (I'm still not sure exactly what), but as I'm working simultaneously with the ATtiny as a (software) master on one bus and a (hardware) slave on another, separate bus (and, there, i.e. in the slave implementation, there are indeed significant differences between the 84 and the 841), I automatically assumed the issue was one of compatibility when things didn't work as expected on the master side. In light of your response, I spent the last couple of days re-building the circuits and slowly going over the code of both the master-side and the slave-side and now it seems to be working perfectly.

Anyway, sorry again to have troubled you with this. At least now you know your library has also been tested and proven to work with the ATtiny841 :-)

All the best,
Nadav

closed 2016-12-08 by felias-fogg