felias-fogg/SoftI2CMaster #17: Fatal error at compile

remygrandin, 2016-12-05:
Hi,

I'm trying to use your library on an arduino due but the compiler don't seam to be on my side ^^

`SoftI2CMaster.h: 63:20: fatal error: avr\io.h: No such file or directory
   #include <avr\io.h>
`

do you have any idea why it won't compile ?

felias-fogg, 2016-12-08:
Hi,
the library is meant to run on ATmega and ATtiny MCUs. For ARM MCU such as the one on the Due, there is no way that the library will work because it is heavily based on using assembler code.

closed 2016-12-08 by felias-fogg

remygrandin, 2016-12-08:
You should probably write that it's not compatible with the due or zero somewhere on the frontpage or on the the playground.arduino.cc page ^^

felias-fogg, 2016-12-08:
It does say: "It can use any pins on any AVR chip to drive the SDA and SCL lines.“ AVR is the ATmega/ATXMega/ATtiny family.  But I should say perhaps explicitly that it doesn’t run on ARMs.

felias-fogg, 2016-12-10:
If you want to have flexibility concerning the pin usage (and this was probably the reason you tried my library, right?), you could actually use the i2cMaster library by landis that I mentioned in the Playground article: https://github.com/landis/arduino/tree/master/libraries/I2cMaster . You have to remove the TwiMaster class, because the hardware registers are named differently. However, then it works beautifully. I tested it on a Genuino Zero and the bus clock is around 100 kHz.