felias-fogg/SoftI2CMaster #18: Doesn't compile for ATMEGA32u4

dHutchings, 2016-12-22:
the Atmega 32u4 is used on Arduino Leonardo & Micros.  I believe it is an AVR chip, which this library supports.

```
The exact stack trace:

/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h: In function 'void i2c_wait_scl_high()':
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:185:31: error: 'SCL_PORT' was not declared in this scope
 #define SCL_IN  (_SFR_IO_ADDR(SCL_PORT) - 2)
                               ^
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:223:23: note: in expansion of macro 'SCL_IN'
      : : [SCLIN] "I" (SCL_IN), [SCLPIN] "I" (SCL_PIN));
                       ^
In file included from /home/douglash/Desktop/double_pendulum/code/integrated/DAC.ino:1:0:
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:223:46: error: 'SCL_PIN' was not declared in this scope
      : : [SCLIN] "I" (SCL_IN), [SCLPIN] "I" (SCL_PIN));
                                              ^
In file included from /root/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/avr/include/avr/io.h:99:0,
                 from /root/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/avr/include/avr/pgmspace.h:90,
                 from /root/.arduino15/packages/arduino/hardware/avr/1.6.15/cores/arduino/Arduino.h:28,
                 from sketch/integrated.ino.cpp:1:
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h: In function 'bool i2c_init()':
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:181:38: error: 'SCL_PORT' was not declared in this scope
 #define SCL_DDR        (_SFR_IO_ADDR(SCL_PORT) - 1)
                                      ^
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:277:23: note: in expansion of macro 'SCL_DDR'
        [SCLDDR] "I"  (SCL_DDR), [SCLPIN] "I" (SCL_PIN), 
                       ^
In file included from /home/douglash/Desktop/double_pendulum/code/integrated/DAC.ino:1:0:
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:277:47: error: 'SCL_PIN' was not declared in this scope
        [SCLDDR] "I"  (SCL_DDR), [SCLPIN] "I" (SCL_PIN), 
                                               ^
In file included from /root/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/avr/include/avr/io.h:99:0,
                 from /root/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/avr/include/avr/pgmspace.h:90,
                 from /root/.arduino15/packages/arduino/hardware/avr/1.6.15/cores/arduino/Arduino.h:28,
                 from sketch/integrated.ino.cpp:1:
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:180:38: error: 'SDA_PORT' was not declared in this scope
 #define SDA_DDR        (_SFR_IO_ADDR(SDA_PORT) - 1)
                                      ^
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:279:23: note: in expansion of macro 'SDA_DDR'
        [SDADDR] "I"  (SDA_DDR), [SDAPIN] "I" (SDA_PIN), 
                       ^
In file included from /home/douglash/Desktop/double_pendulum/code/integrated/DAC.ino:1:0:
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:279:47: error: 'SDA_PIN' was not declared in this scope
        [SDADDR] "I"  (SDA_DDR), [SDAPIN] "I" (SDA_PIN), 
                                               ^
In file included from /root/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/avr/include/avr/io.h:99:0,
                 from /root/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/avr/include/avr/pgmspace.h:90,
                 from /root/.arduino15/packages/arduino/hardware/avr/1.6.15/cores/arduino/Arduino.h:28,
                 from sketch/integrated.ino.cpp:1:
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h: In function 'bool i2c_start(uint8_t)':
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:180:38: error: 'SDA_PORT' was not declared in this scope
 #define SDA_DDR        (_SFR_IO_ADDR(SDA_PORT) - 1)
                                      ^
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:297:25: note: in expansion of macro 'SDA_DDR'
      : : [SDADDR] "I"  (SDA_DDR), [SDAPIN] "I" (SDA_PIN),
                         ^
In file included from /home/douglash/Desktop/double_pendulum/code/integrated/DAC.ino:1:0:
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:297:49: error: 'SDA_PIN' was not declared in this scope
      : : [SDADDR] "I"  (SDA_DDR), [SDAPIN] "I" (SDA_PIN),
                                                 ^
In file included from /root/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/avr/include/avr/io.h:99:0,
                 from /root/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/avr/include/avr/pgmspace.h:90,
                 from /root/.arduino15/packages/arduino/hardware/avr/1.6.15/cores/arduino/Arduino.h:28,
                 from sketch/integrated.ino.cpp:1:
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:185:31: error: 'SCL_PORT' was not declared in this scope
 #define SCL_IN  (_SFR_IO_ADDR(SCL_PORT) - 2)
                               ^
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:298:21: note: in expansion of macro 'SCL_IN'
        [SCLIN] "I" (SCL_IN),[SCLPIN] "I" (SCL_PIN)); 
                     ^
[... 128 lines cut ...]
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:515:22: note: in expansion of macro 'SCL_DDR'
       [SCLDDR] "I"  (SCL_DDR), [SCLPIN] "I" (SCL_PIN), [SCLIN] "I" (SCL_IN),
                      ^
In file included from /home/douglash/Desktop/double_pendulum/code/integrated/DAC.ino:1:0:
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:515:46: error: 'SCL_PIN' was not declared in this scope
       [SCLDDR] "I"  (SCL_DDR), [SCLPIN] "I" (SCL_PIN), [SCLIN] "I" (SCL_IN),
                                              ^
In file included from /root/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/avr/include/avr/io.h:99:0,
                 from /root/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/avr/include/avr/pgmspace.h:90,
                 from /root/.arduino15/packages/arduino/hardware/avr/1.6.15/cores/arduino/Arduino.h:28,
                 from sketch/integrated.ino.cpp:1:
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:180:38: error: 'SDA_PORT' was not declared in this scope
 #define SDA_DDR        (_SFR_IO_ADDR(SDA_PORT) - 1)
                                      ^
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:516:22: note: in expansion of macro 'SDA_DDR'
       [SDADDR] "I"  (SDA_DDR), [SDAPIN] "I" (SDA_PIN), [SDAIN] "I" (SDA_IN) 
                      ^
In file included from /home/douglash/Desktop/double_pendulum/code/integrated/DAC.ino:1:0:
/root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:516:46: error: 'SDA_PIN' was not declared in this scope
       [SDADDR] "I"  (SDA_DDR), [SDAPIN] "I" (SDA_PIN), [SDAIN] "I" (SDA_IN) 
                                              ^
exit status 1
Error compiling for board Arduino/Genuino Micro.

```

dHutchings, 2016-12-22:
I'd just like to follow up.

I was able to work around my issues by using your soft-slow libraries.  Thank you for making these libraries!  They were a great help.

felias-fogg, 2016-12-22:
Hi Doug,

the error messages say that you haven’t declared SCL_PORT, SDA_PORT, SCL_PIN, and SDA_PIN. As described in the the playground, you need to declare these constants before importing the library.

> /root/Arduino/libraries/SoftI2CMaster-master/SoftI2CMaster.h:185:31: error: 'SCL_PORT' was not declared in this scope
>  #define SCL_IN  (_SFR_IO_ADDR(SCL_PORT) - 2)
>                                ^

closed 2016-12-23 by felias-fogg