felias-fogg/SoftI2CMaster #19: How do I use multiple instances?

Humancell, 2017-02-28:
I have a project where I want to use 2-3 instances of this to talk over 2-3 different I2C busses at the same time.

How can I do this with this library?  Or can I?

For now, I made a complete copy of the library, renamed it, and renamed all of the functions and defines ... as a work-around.  Is there a better way?

felias-fogg, 2017-02-28:
Hi Scott,

no, there is no better way. However, you can use the SlowSoftI2CMaster library, which is entirely written in C++, which allows for multiple instances: https://github.com/felias-fogg/SlowSoftI2CMaster . 

Cheers,
Bernhard

Humancell, 2017-03-01:
Thanks for the reply.  And thanks for the pointer to the other library.  I'll continue to see what I can do, and will use my "renamed" versions also.  Thanks for all of this code ... it's a wonderful contribution!

closed 2017-03-01 by Humancell