felias-fogg/SoftI2CMaster #20: SoftI2CMaster with internal pull-up

harusametime, 2017-03-07:
Hi, I am developing an Arduino program for ATmega328 to communicate with sensors via I2C lines that are NOT pulled up. Differently from Arduino Wire Library and your SlowSoftI2CMaster, SoftI2CMaster does not support internal pull-up. Now I am trying to change SoftI2CMaster.h for enabling the internal pull-up.

According to p. 99 in the datasheet of ATmega328, 
http://www.atmel.com/Images/Atmel-42735-8-bit-AVR-Microcontroller-ATmega328-328P_Datasheet.pdf
`{DDxn, PORTxn} = 0b01` is the pull-up enabled state. As for `bool i2c_init(void)`, I change the header file as follows:
` " cbi      %[SDAOUT],%[SDAPIN]     ;clear SDA output value \n\t"`
` " cbi      %[SCLOUT],%[SCLPIN]     ;clear SCL output value \n\t" `
is changed to
`" sbi      %[SDAOUT],%[SDAPIN]     ;" `
` " sbi      %[SCLOUT],%[SCLPIN]     ;" `
for  PORTxn =1. I suppose this works fine.

However, I have no idea of how to change the other functions, such as `bool  i2c_start(uint8_t addr)`.
Could you help me to understand how to change the header file? Or, does SlowSoftI2CMaster work as fast as SoftI2CMaster?

felias-fogg, 2017-03-23:
Hi,

I recommend to use the SlowSoftI2CMaster lib. It runs with approx. 40kHz on an UNO board (16MHz), which is enuogh for most purposes. 

Best,
Bernhard

harusametime, 2017-03-23:
Thank you for your reply.  I will try SlowSoftI2CMaster lib.

closed 2017-03-23 by harusametime

felias-fogg, 2018-02-04:
Well, meanwhile I have added a compiler option to use the internal pullups (this is unsafe!)

harusametime, 2018-02-05:
A great update!
My project now goes well with Wire library, but want to try it for the time when I want faster I2C speed.