felias-fogg/SoftI2CMaster #24: Issue to setup I2C with Esplora Board

laenan8466, 2018-02-14:
Hi,
awesome project! 
I'm trying to get it working on my Arduino Esplora Board, with the TinkerKit outputs, till now to now success.

These are located on D11 & D3 ([Reference](http://21stdigitalhome.blogspot.de/2013/01/arduino-esplora-tinkerkit-outputs.html)). The Esplora is used as Master, an Mega as slave.

Minimal compiling example for the Esplora is:
```
#include <Esplora.h>
#include <SPI.h>

// Setup I2C connection
#define SDA_PORT PORTB
#define SDA_PIN 3 // = D11
#define SCL_PORT PORTD
#define SCL_PIN 3 // = D3
#define I2C_SLOWMODE 1
 
#include "SoftWire.h" // .h is in sketch-folder, #include "SoftI2CMaster.h" in SoftWire.h adjusted

#define I2C_7BITADDR 0x68 // DS1307
#define MEMLOC 0x0A
#define ADDRLEN 1

SoftWire Wire = SoftWire();

void setup() {
  Serial.begin(57600);
  Wire.begin();

}

void loop() {
  Wire.beginTransmission(I2C_7BITADDR);
  for (byte i=1; i<ADDRLEN; i++) Wire.write(0x00);
  Wire.write(MEMLOC);
  Wire.endTransmission(false);
  Wire.requestFrom(I2C_7BITADDR,1);
  byte val = Wire.read();
  Serial.println(val);

}
```

And on Mega
```
#define I2C_7BITADDR 0x68 // DS1307
#define MEMLOC 0x0A 

#include <Wire.h>

#define SERIAL_BAUD_RATE  57600

void setup()
{
  Serial.begin(SERIAL_BAUD_RATE);
  delay(1000);  // Wait for 1 second before starting the program.
  Wire.begin(I2C_7BITADDR);
  Wire.onReceive(receiveEvent);

}
 
 
void loop()
{
   delay(100);
}

// function that executes whenever data is received from master
// this function is registered as an event, see setup()
void receiveEvent(int howMany) {
  byte c = Wire.read(); // receive byte
  Serial.print(c);         // print the byte
}
```

But on the Mega no Serial output is received. 
Has anyone ever tried to make it happen, with an Esplora board? Any suggestions how to debug? I would have an oscilloscope availible, but not too eager to power it up. ;-)

Best regards,
Laenan

felias-fogg, 2018-02-14:
Hi,

Esplora should work, but I never tried one of those. One common problem is to forget the pullups! You need to declare
#define I2C_PULLUP 1
Even better: Use external pullup resistors, 2.2k or 4.7k

laenan8466, 2018-02-17:
So hey,
sorry, it took me some time, to make further checks. So it was a combined problem:

- First, the pullups, like you pointed out.
- Second, the PinLayout is different other Microcontrollers, so one has to use:
```
#define SDA_PORT PORTB
#define SDA_PIN 7 // = D11
#define SCL_PORT PORTD
#define SCL_PIN 0 // = D3
```

So thank you again and maybe this will help others to make it work with the Esplora.

closed 2018-02-17 by laenan8466