felias-fogg/SoftI2CMaster #27: util/twi.h: No such file or directory

parkerlreed, 2018-03-11:
````
In file included from /home/parker/Pictures/PackProbe/PackProbe.ino:54:0:
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:81:22: fatal error: util/twi.h: No such file or directory
compilation terminated.
Error compiling for board Teensy 3.6.
````

````
[parker@e55 ~]$ find /usr/share/arduino/ -name "*twi.h"
/usr/share/arduino/hardware/teensy/avr/libraries/Wire/utility/twi.h
/usr/share/arduino/hardware/archlinux-arduino/avr/libraries/Wire/src/utility/twi.h
````

I tried changing the twi.h include to the absolute path

````
#include </usr/share/arduino/hardware/teensy/avr/libraries/Wire/utility/twi.h>
````

But that ends up with this

````
In file included from /home/parker/Pictures/PackProbe/PackProbe.ino:54:0:
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h: In function 'void i2c_wait_scl_high()':
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:228:39: error: '_SFR_IO_ADDR' was not declared in this scope
 #define SCL_IN  (_SFR_IO_ADDR(SCL_PORT) - 2)
                                       ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:265:23: note: in expansion of macro 'SCL_IN'
      : : [SCLIN] "I" (SCL_IN), [SCLPIN] "I" (SCL_PIN));
                       ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h: In function 'bool i2c_init()':
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:224:46: error: '_SFR_IO_ADDR' was not declared in this scope
 #define SCL_DDR        (_SFR_IO_ADDR(SCL_PORT) - 1)
                                              ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:346:23: note: in expansion of macro 'SCL_DDR'
        [SCLDDR] "I"  (SCL_DDR), [SCLPIN] "I" (SCL_PIN), 
                       ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h: In function 'bool i2c_start(uint8_t)':
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:223:46: error: '_SFR_IO_ADDR' was not declared in this scope
 #define SDA_DDR        (_SFR_IO_ADDR(SDA_PORT) - 1)
                                              ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:409:25: note: in expansion of macro 'SDA_DDR'
      : : [SDADDR] "I"  (SDA_DDR), [SDAPIN] "I" (SDA_PIN),
                         ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h: In function 'bool i2c_rep_start(uint8_t)':
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:224:46: error: '_SFR_IO_ADDR' was not declared in this scope
 #define SCL_DDR        (_SFR_IO_ADDR(SCL_PORT) - 1)
                                              ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:453:25: note: in expansion of macro 'SCL_DDR'
      : : [SCLDDR] "I"  (SCL_DDR), [SCLPIN] "I" (SCL_PIN),
                         ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h: In function 'bool i2c_start_wait(uint8_t)':
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:223:46: error: '_SFR_IO_ADDR' was not declared in this scope
 #define SDA_DDR        (_SFR_IO_ADDR(SDA_PORT) - 1)
                                              ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:554:24: note: in expansion of macro 'SDA_DDR'
     : : [SDADDR] "I"  (SDA_DDR), [SDAPIN] "I" (SDA_PIN), [SDAOUT] "I" (SDA_OUT),
                        ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:559:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h: In function 'void i2c_stop()':
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:224:46: error: '_SFR_IO_ADDR' was not declared in this scope
 #define SCL_DDR        (_SFR_IO_ADDR(SCL_PORT) - 1)
                                              ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:606:25: note: in expansion of macro 'SCL_DDR'
      : : [SCLDDR] "I"  (SCL_DDR), [SCLPIN] "I" (SCL_PIN), [SCLIN] "I" (SCL_IN),
                         ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h: In function 'bool i2c_write(uint8_t)':
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:224:46: error: '_SFR_IO_ADDR' was not declared in this scope
 #define SCL_DDR        (_SFR_IO_ADDR(SCL_PORT) - 1)
                                              ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:734:22: note: in expansion of macro 'SCL_DDR'
       [SCLDDR] "I"  (SCL_DDR), [SCLPIN] "I" (SCL_PIN), [SCLIN] "I" (SCL_IN),
                      ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h: In function 'uint8_t i2c_read(bool)':
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:224:46: error: '_SFR_IO_ADDR' was not declared in this scope
 #define SCL_DDR        (_SFR_IO_ADDR(SCL_PORT) - 1)
                                              ^
/home/parker/Arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:844:22: note: in expansion of macro 'SCL_DDR'
       [SCLDDR] "I"  (SCL_DDR), [SCLPIN] "I" (SCL_PIN), [SCLIN] "I" (SCL_IN),
                      ^
Error compiling for board Teensy 3.6.
````

I'm able to select a regular Arduino board and the sketch compiles fine, so it my be something with the Teensy libraries but I';m not sure what.

felias-fogg, 2018-03-11:
Yes, it has to do with the Teensy 3.6 board. This board does not have an AVR MCU, but an ARM MCU. Please read the third paragraph of the README.md file, starting with 

> If you want a solution running on an ARM MCU (Due, Zero, Teensy 3.x), ...

closed 2018-03-11 by felias-fogg

parkerlreed, 2018-03-11:
Any suggestions on converting the existing code?

Trying SlowSoftI2CMaster.h in place of SoftI2CMaster.h in the include gives me

````
/usr/share/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -tools /usr/share/arduino/tools-builder -tools /usr/share/arduino/hardware/tools/avr -libraries /home/parker/Arduino/libraries -fqbn=teensy:avr:teensy36:usb=serial,speed=180,opt=o2std,keys=en-us -ide-version=10805 -build-path /tmp/arduino_build_343537 -warnings=none -build-cache /tmp/arduino_cache_879247 -verbose /home/parker/Pictures/PackProbe/PackProbe.ino
/usr/share/arduino/arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -tools /usr/share/arduino/tools-builder -tools /usr/share/arduino/hardware/tools/avr -libraries /home/parker/Arduino/libraries -fqbn=teensy:avr:teensy36:usb=serial,speed=180,opt=o2std,keys=en-us -ide-version=10805 -build-path /tmp/arduino_build_343537 -warnings=none -build-cache /tmp/arduino_cache_879247 -verbose /home/parker/Pictures/PackProbe/PackProbe.ino
Using board 'teensy36' from platform in folder: /usr/share/arduino/hardware/teensy/avr
Using core 'teensy3' from platform in folder: /usr/share/arduino/hardware/teensy/avr
Detecting libraries used...
"/usr/share/arduino/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++14 -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=141 -DARDUINO=10805 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-I/usr/share/arduino/hardware/teensy/avr/cores/teensy3" "/tmp/arduino_build_343537/sketch/PackProbe.ino.cpp" -o "/dev/null"
"/usr/share/arduino/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++14 -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=141 -DARDUINO=10805 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-I/usr/share/arduino/hardware/teensy/avr/cores/teensy3" "-I/home/parker/Arduino/libraries/SlowSoftI2CMaster" "/tmp/arduino_build_343537/sketch/PackProbe.ino.cpp" -o "/dev/null"
"/usr/share/arduino/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++14 -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=141 -DARDUINO=10805 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-I/usr/share/arduino/hardware/teensy/avr/cores/teensy3" "-I/home/parker/Arduino/libraries/SlowSoftI2CMaster" "/home/parker/Arduino/libraries/SlowSoftI2CMaster/SlowSoftI2CMaster.cpp" -o "/dev/null"
Generating function prototypes...
"/usr/share/arduino/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++14 -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=141 -DARDUINO=10805 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-I/usr/share/arduino/hardware/teensy/avr/cores/teensy3" "-I/home/parker/Arduino/libraries/SlowSoftI2CMaster" "/tmp/arduino_build_343537/sketch/PackProbe.ino.cpp" -o "/tmp/arduino_build_343537/preproc/ctags_target_for_gcc_minus_e.cpp"
"/usr/bin/arduino-ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino_build_343537/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/usr/share/arduino/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++" -c -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -fno-exceptions -felide-constructors -std=gnu++14 -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=141 -DARDUINO=10805 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-I/usr/share/arduino/hardware/teensy/avr/cores/teensy3" "-I/home/parker/Arduino/libraries/SlowSoftI2CMaster" "/tmp/arduino_build_343537/sketch/PackProbe.ino.cpp" -o "/tmp/arduino_build_343537/sketch/PackProbe.ino.cpp.o"
PackProbe: In function 'void setup()':
PackProbe:133: error: 'i2c_init' was not declared in this scope
   Serial.println(i2c_init());
                           ^
PackProbe: In function 'int fetchWord(byte)':
PackProbe:151: error: 'i2c_start' was not declared in this scope
   i2c_start(deviceAddress << 1 | I2C_WRITE);
                                           ^
PackProbe:152: error: 'i2c_write' was not declared in this scope
   i2c_write(func);
                 ^
PackProbe:153: error: 'i2c_rep_start' was not declared in this scope
   i2c_rep_start(deviceAddress << 1 | I2C_READ);
                                              ^
PackProbe:154: error: 'i2c_read' was not declared in this scope
   byte b1 = i2c_read(false);
                           ^
PackProbe:156: error: 'i2c_stop' was not declared in this scope
   i2c_stop();
            ^
PackProbe: In function 'uint8_t i2c_smbus_read_block(uint8_t, uint8_t*, uint8_t)':
PackProbe:163: error: 'i2c_start' was not declared in this scope
   i2c_start((deviceAddress << 1) + I2C_WRITE);
                                             ^
[... cut ...]
  Serial.print("Battery Status (BIN): 0b");
  Serial.println(fetchWord(BATTERY_STATUS), BIN);

  Serial.print("Charging Current: ");
  Serial.println(fetchWord(CHARGING_CURRENT));

  Serial.print("Charging Voltage: ");
  Serial.println(fetchWord(CHARGING_VOLTAGE));

  Serial.print("Temp: ");
  unsigned int tempk = fetchWord(TEMPERATURE);
  Serial.println((float)tempk / 10.0 - 273.15);

  Serial.print("Current (mA): " );
  Serial.println(fetchWord(CURRENT));

  Serial.println(".");
  delay(1000);
}
````

felias-fogg, 2018-03-12:
Yes, SlowSoftI2CMaster is not a drop-in replacement. It defines a class, so you have to use a slightly different interface. Please read the reader file and study the examples.