felias-fogg/SoftI2CMaster #29: Readme Documentation Issue: i2c_start

frozenspongebob, 2018-03-15:
Hello,

I just wanted to report a slight but potentially significant discrepancy between the Readme documentation and actual results. In the documentation, in the explanation of i2c_start(), it states

> `i2c_start(addr)`
Initiates a transfer to the slave device with the 8-bit I2C address addr. Note that this library uses the 8-bit addressing scheme different from the 7-bit scheme in the Wire library. In addition the least significant bit of addr must be specified as I2C_WRITE (=0) or I2C_READ (=1). Returns **true** if the addressed device replies with an ACK. Otherwise **false** is returned.

Assuming true is 1 and false is 0, I expected a 1 when the master **was** receiving a response from the slave and 0 otherwise. However, in actual testing, I discovered that when the master was receiving a response, i2c_start() returned **0**! Although it may not be a potentially project-threatening issue, I just wanted to point that out. Or, I could just be setting it up wrong. Who knows?

Thanks

felias-fogg, 2018-03-17:
Well, I am not sure what you did. But when you look at the example sketch I2CScanSoft, you'll see that 1/true is used as the indicator that an I2C device responds with an ACK. And this sketch has worked so far without a problem.

frozenspongebob, 2018-03-18:
Take a look: I think I followed your documentation correctly, but I do not get a response when I try with 1.

https://pastebin.com/nsqKXiDN

felias-fogg, 2018-03-18:
If you substitute the call to I2c_write with the intended call to i2c_start in line 33, then everything should work.

closed 2018-03-20 by felias-fogg