felias-fogg/SoftI2CMaster #37: available returns a number even when no device is attached

lpatiny, 2018-09-08:
Thanks for your very useful library !

I'm using SoftWire with the following flags on a home made Leonardo Lilipad Pro (Atmega 32u4 at 8MHz):
```
#define I2C_HARDWARE 1
#define I2C_TIMEOUT  10
#define I2C_SLOWMODE 1
```

When I use the following code:
```
Wire.requestFrom(address, 2);
Serial.println(Wire.available());
```

I always have the value '2' even when the device is not connected. I would expect '0' if nothing is return (and if the device is not connected) which is the result obtained with the native Wire library.

felias-fogg, 2021-02-05:
I now got a "Leonardo"-alike and will check that.

felias-fogg, 2022-02-19:
Well, this should be fixed with the 2.1.9

closed 2022-02-19 by felias-fogg