felias-fogg/SoftI2CMaster #40: Multiple definition of ...

fragsalat, 2019-02-11:
Hey there, I'm having the issue with latest master. 
Using gcc-avr 5.4.0+Atmel3.6.1-1~rpt1 and Arduino 1.0.5+dfsg2-4.1 on raspbian.
I want to build for an ATtiny84 but that should not matter right now. 

That's how I use it
```
#define SDA_PORT PORTA
#define SDA_PIN 4 // = A4
#define SCL_PORT PORTA
#define SCL_PIN 5 // = A5

#include <SoftWire.h>

SoftWire TinyWireM = SoftWire();
```

To be honest the error is a bit confusing as it sounds more less as the file was included twice as the line matches the constructor of SoftWire which doens't declare the functions.
```
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `ass_i2c_delay_half'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `ass_i2c_wait_scl_high'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `i2c_init()'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `i2c_start(unsigned char)'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `ass_i2c_write'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `i2c_rep_start(unsigned char)'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `i2c_start_wait(unsigned char)'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `ass_i2c_stop'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `i2c_read(bool)'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
CCS811/ccs811.cpp.o: In function `_GLOBAL__sub_I_ass_i2c_delay_half':
/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: multiple definition of `TinyWireM'
Blink.cpp.o:/root/sketchbook/libraries/SoftI2CMaster/SoftWire.h:46: first defined here
collect2: error: ld returned 1 exit status
```

Any idea what this can be?

felias-fogg, 2019-02-12:
As written in the description: You are allowed to import SoftWire or SoftI2CMaster only once!

closed 2019-02-12 by felias-fogg

fragsalat, 2019-02-16:
Yeah It actually is just a strange behavior of Arduino IDE. It works when including once in the library CCS811.cpp but it must be a relative import so that Arduino IDE recognize it without import in main .ino file