felias-fogg/SoftI2CMaster #50: Relocation truncated errors

ondrejj, 2020-03-09:
Using softi2c library on Fedora, which has newer gcc and utils, but looks like I can't use SoftI2CMaster anymore without errors. Here is error message:

```
/home/ondrejj/git/arduino/libraries/SoftI2CMaster/SoftI2CMaster.h: In function 'bool i2c_start_wait(uint8_t)':
/home/ondrejj/git/arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:566:1: warning: no return statement in function returning non-void [-Wreturn-type]
  566 | }
      | ^
/tmp/sensor.ino.elf.5VjqRY.ltrans0.ltrans.o: in function `i2c_read(bool)':
/home/ondrejj/git/arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:764:(.text+0x2688): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
/home/ondrejj/git/arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:764:(.text+0x268c): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
/home/ondrejj/git/arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:764:(.text+0x2698): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
/home/ondrejj/git/arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:764:(.text+0x26b8): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
/home/ondrejj/git/arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:764:(.text+0x26c4): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
/home/ondrejj/git/arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:764:(.text+0x26c6): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
/tmp/sensor.ino.elf.5VjqRY.ltrans0.ltrans.o: in function `*ass_i2c_write':
/home/ondrejj/git/arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:646:(.text+0x26f6): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
/home/ondrejj/git/arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:646:(.text+0x2704): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
/home/ondrejj/git/arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:646:(.text+0x270c): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
/home/ondrejj/git/arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:646:(.text+0x2718): relocation truncated to fit: R_AVR_13_PCREL against `no symbol'
/home/ondrejj/git/arduino/libraries/SoftI2CMaster/SoftI2CMaster.h:646:(.text+0x2726): additional relocation overflows omitted from the output
collect2: error: ld returned 1 exit status
```

Some more about this here:
https://www.avrfreaks.net/forum/linking-error-relocation-truncated-fit-ravr13pcrel-against-no-symbol

Looks like changing rjmp=>jmp and rcall=>call works as workaround. Can you please fix it upstream?

Gabrielerusso, 2021-03-04:
i solved it by replacing RCALL and RJMP with the "bigger" CALL and JMP in the library source file, looks a common problem

trullock, 2021-08-16:
This issue looks like it can be closed?

felias-fogg, 2021-08-16:
Indeed!

closed 2021-08-16 by felias-fogg