felias-fogg/SoftI2CMaster #53: Problems with Newer Arduino IDE Version

oetkenSG, 2020-05-27:
We have experienced consistent problems when using newer versions of the Arduino IDE.

On Arduino IDE version 1.8.12, we have not been able to communicate over I2C.  The code compiles and runs, but the communication itself seems to fail at runtime. We tested on multiple different machines on both macOS Catalina and Windows 10.

Reverting to Arduino IDE version 1.8.9 fixed the issue.  This is not ideal for us since this version of the IDE is 32-bit which is no longer supported on macOS.

We would very much appreciate your help on this.

Thank you for your hard work!

rakeshpatil1983, 2020-09-13:
Any update on this i'm having version 1.8.13 and it does not work. some other libraries support 1.8.13 
Thanks.

felias-fogg, 2021-02-04:
Sorry for the late answer. I am using 1.8.13 and do not encounter any problems. So what is your particular problem? Can you send me the sketch that does not work?

closed 2021-11-11 by felias-fogg