felias-fogg/SoftI2CMaster #54: Example Eeprom24AA1025softI2C not able to write values from D0 to D9

x225am, 2021-01-30:
Hi, I tried to play with example code and everything works fine, except when I trying to write value via terminal to the eeprom. For some reason when I am trying to write values from D0 to D9 then it only writing first byte 'D' and then showing 'Unknown command.
This what I am sending:
w2#D9#
That's what receiving:
Write to addr: 2
Value: D
Writing...
Unknown command

When I am reading back
r2#
It reads only 0D
Read from addr: 2
Reading...
2: 0D

I tried all other values from 00 to FF all works except D0-D9.
also I just tried to write direct byte D9 on setup function and it works without issue.
Anyone have any idea how to solve issue with writing 'DX' bytes

x225am, 2021-01-30:
I believe there are issue in the next function:
unsigned long parseHex() {
  unsigned long result = 0L;
  char byte = '\0';

  //  while (hexdigit(byte)) {
  while (byte != '\r' && byte != '#') {
    while (!Serial.available());
    byte = Serial.read();
    // if (!hexdigit(byte)) break;
    if (byte == '\r' || byte == '#') break;
    if (byte >= 'a' && byte <= 'f')
      byte = byte -'a' + 'A';
    if ((byte >= '0' && byte <= '9') ||
	(byte >= 'A' && byte <= 'F')) {
      Serial.print(byte);
      if (byte >= '0' && byte <= '9') byte = byte - '0';
      else byte = byte - 'A' + 10;
      result = result * 16;
      result = result + byte;
    }
  }
  Serial.println();
  return result;
}

felias-fogg, 2021-01-31:
You are right! I fixed the bug.

closed 2021-02-01 by felias-fogg

x225am, 2021-02-04:
Hi thanks you to fixing it. However it still not working when you trying to read from address 0 it reads from address 30
Read from addr: 0
Reading...
30: 40

felias-fogg, 2021-02-05:
Sorry, another stupid bug. Also fixed.