felias-fogg/SoftI2CMaster #61: What is the reason that this awesome library is not included in the Arduino library manager.

ArminJo, 2021-11-05:
Hallo Bernhard,
do you need support for this? I can help / do it for you.
Best regards
Armin

felias-fogg, 2021-11-05:
Hi Armin,

would be great! I am currently busy with a lot of other stuff. Would you create a PR?

Best,
Bernhard

ArminJo, 2021-11-10:
Any comments? Too many changes?
Best regards
Armin

felias-fogg, 2021-11-10:
OK, done. You said, you are now requesting to include it as an Arduino lib?

The CI stuff is amazing. Is there a tutorial somewhere? And if I understood right, it all works on GitHub, right? I do not have to setup anything.

Best,
Bernhard

ArminJo, 2021-11-10:
Is [this ](https://github.com/ArminJo/arduino-test-compile)a tutorial?
Please give me feedback.
And yes it is intended for the lazy programmer...

The library is now registered for Arduino https://github.com/arduino/library-registry/pull/658, do not know how long it takes to have it avalible by the Arduino IDE.

closed 2021-11-10 by ArminJo

felias-fogg, 2021-11-11:
Well, I guess I wouldn't call it a tutorial. But it is good enough for me to find out what to do next time when I create a library or project. Thanks!