felias-fogg/SoftI2CMaster #75: Using SoftWire with Wire library spawns error

sandric, 2022-07-23:
Hi, I tried to use `SoftWire.h` wrapper in my code, together with `real` i2c `Wire.h` library as a slave - the idea there is that I need to read slave i2c devices (with software i2c master) with arduino, and then send computed result to raspberrypi with slave hardware i2c line. But I have an error at compilation:

```
In file included from /home/sandric/Arduino/soft_i2c_test/soft_i2c_test.ino:19:0:
/home/sandric/.arduino15/packages/arduino/hardware/avr/1.8.5/libraries/Wire/src/Wire.h:86:16: error: conflicting declaration 'TwoWire Wire'
 extern TwoWire Wire;
                ^~~~
In file included from /home/sandric/Arduino/soft_i2c_test/soft_i2c_test.ino:18:0:
/home/sandric/Arduino/libraries/SoftI2CMaster/src/SoftWire.h:248:10: note: previous declaration as 'SoftWire Wire'
 SoftWire Wire = SoftWire();
          ^~~~
exit status 1
Error compiling for board Arduino Uno.
```

Below is my code:

```
#define SDA_PORT PORTD
#define SDA_PIN 7

#define SCL_PORT PORTD
#define SCL_PIN 6

#define I2C_FASTMODE 1
#define I2C_TIMEOUT 100
#define I2C_FASTMODE 1
     
#include <SoftWire.h>
#include <Wire.h>

SoftWire SWire = SoftWire();

void setup()
{
  SWire.begin();
  
  Serial.begin(9600);
  while (!Serial);             // Leonardo: wait for serial monitor
  Serial.println("\nI2C Scanner ssss");
}

void loop() {}
```

Obviously the issue that Wire is redefined in SoftWIre.h - I could remove it myself changing library on my machine, but thought mb there's better way to do this?

felias-fogg, 2022-07-29:
Hi,

> Obviously the issue that Wire is redefined in SoftWIre.h - I could remove it myself changing library on my machine, but thought mb there's better way to do this?

No, that would be the only way out. However, you could decide to use a different communication mechanism to talk with the raspi, e.g., serial communication. 

Best,
Bernhard

sandric, 2022-07-30:
Yeah, I redefined and all works great. As for uart - I left only 1 software i2c after DPI screen took all but 6 gpio's hehe, but it works with redefenition anyways.

closed 2022-07-30 by sandric