felias-fogg/SoftI2CMaster #78: Problem using an Arduino Mega 2560

rauhur, 2023-04-11:
When I configure the I2C pins for an Arduino Mega 2560:

```
#define SCL_PIN 21              //Default SDA is Pin5 PORTC for the UNO -- you can set this to any tristate pin
#define SCL_PORT PORTD 
#define SDA_PIN 20              //Default SCL is Pin4 PORTC for the UNO -- you can set this to any tristate pin
#define SDA_PORT PORTD
```

I receive an assembler compilation error:

```
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s: Assembler messages:
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1465: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1466: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1471: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1477: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1481: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1487: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1490: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1493: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1499: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1503: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1509: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1544: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1550: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1553: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1557: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1562: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1573: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1576: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1580: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1584: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1587: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1593: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1619: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1620: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1622: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1624: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1626: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1657: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1659: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1696: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1698: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1700: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1702: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1704: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1732: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1734: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1834: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1835: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1836: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1837: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1840: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:1842: Error: number must be positive and less than 8
C:\Users\user_A\AppData\Local\Temp\cccOXI9H.s:2571: Error: number must be positive and less than 8
lto-wrapper.exe: fatal error: C:\Users\user_A\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc returned 1 exit status
compilation terminated.
c:/users/user_A/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1r: C:\Users\rauhur\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc returned 1 exit status
compilation terminated.
c:/users/rauhur/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

Am I making a mistake in the configuration or is it a bug in the library?

felias-fogg, 2023-04-11:
Hi, this is a config error. You have to specify the pin in port register D, not the Arduino pin number. Digital pin 20 (SDA) on an Arduino Mega is D1, pin 21 (SCL) is D0. So the spec should be
#define SCL_PIN 0              
#define SCL_PORT PORTD 
#define SDA_PIN 1  
#define SDA_PORT PORTD

closed 2023-04-11 by felias-fogg

rauhur, 2023-04-12:
Thanks for your help, it works fine now.